doe300/VC4CL #100: mbox_property Connection timed out error

xuelian-bai, 2020-11-09:
Hi, I have ported VC4CL to Tizen on RPI3, but I met a problem. When I run sample app, there is error:
`
[VC4CL](      simple_cl): API call: void* clGetExtensionFunctionAddressForPlatform(cl_platform_id 0x1e3b874, const char* "clIcdGetPlatformIDsKHR")
[VC4CL](      simple_cl): get extension function address: clIcdGetPlatformIDsKHR
[VC4CL](      simple_cl): API call: void* clGetExtensionFunctionAddressForPlatform(cl_platform_id 0x1e3b874, const char* "clGetPlatformInfo")
[VC4CL](      simple_cl): get extension function address: clGetPlatformInfo
[VC4CL](      simple_cl): API call: cl_int clIcdGetPlatformIDsKHR(cl_uint 0, cl_platform_id* 0, cl_uint* 0x7e864308)
[VC4CL](      simple_cl): API call: cl_int clIcdGetPlatformIDsKHR(cl_uint 1, cl_platform_id* 0x1e31778, cl_uint* 0)
[VC4CL](      simple_cl): API call: cl_int clGetPlatformInfo(cl_platform_id 0x1e3b874, cl_platform_info 2308, size_t 0, void* 0, size_t* 0x7e864280)
[VC4CL](      simple_cl): API call: cl_int clGetPlatformInfo(cl_platform_id 0x1e3b874, cl_platform_info 2308, size_t 226, void* 0x1e40cc8, size_t* 0)
[VC4CL](      simple_cl): API call: cl_int clGetPlatformInfo(cl_platform_id 0x1e3b874, cl_platform_info 2336, size_t 0, void* 0, size_t* 0x7e864280)
[VC4CL](      simple_cl): API call: cl_int clGetPlatformInfo(cl_platform_id 0x1e3b874, cl_platform_info 2336, size_t 6, void* 0x1e36a18, size_t* 0)
[VC4CL](      simple_cl): API call: cl_int clGetPlatformInfo(cl_platform_id 0x1e3b874, cl_platform_info 2304, size_t 0, void* 0, size_t* 0x7e864280)
[VC4CL](      simple_cl): API call: cl_int clGetPlatformInfo(cl_platform_id 0x1e3b874, cl_platform_info 2304, size_t 17, void* 0x1e3fac0, size_t* 0)
[VC4CL](      simple_cl): API call: cl_int clGetPlatformInfo(cl_platform_id 0x1e3b874, cl_platform_info 2305, size_t 0, void* 0, size_t* 0x7e864280)
[VC4CL](      simple_cl): API call: cl_int clGetPlatformInfo(cl_platform_id 0x1e3b874, cl_platform_info 2305, size_t 36, void* 0x1e3f2e8, size_t* 0)
[VC4CL](      simple_cl): API call: cl_int clGetPlatformInfo(cl_platform_id 0x1e3b874, cl_platform_info 2306, size_t 0, void* 0, size_t* 0x7e864280)
[VC4CL](      simple_cl): API call: cl_int clGetPlatformInfo(cl_platform_id 0x1e3b874, cl_platform_info 2306, size_t 45, void* 0x1e40108, size_t* 0)
[VC4CL](      simple_cl): API call: cl_int clGetPlatformInfo(cl_platform_id 0x1e3b874, cl_platform_info 2307, size_t 0, void* 0, size_t* 0x7e864280)
[VC4CL](      simple_cl): API call: cl_int clGetPlatformInfo(cl_platform_id 0x1e3b874, cl_platform_info 2307, size_t 7, void* 0x1e3d998, size_t* 0)
[VC4CL](      simple_cl): API call: cl_int clGetDeviceIDs(cl_platform_id 0x1e3b874, cl_device_type 1, cl_uint 1, cl_device_id* 0x7e864498, cl_uint* 0x7e86448c)
1:ret[0], num_platforms[1], num_devices[1]
[VC4CL](      simple_cl): API call: cl_context clCreateContext(const cl_context_properties* 0, cl_uint 1, const cl_device_id* 0x7e864498, void(CL_CALLBACK*)(const char* errinfo,
const void* private_info, size_t cb, void* user_data) 0x7e864354, void* 0, cl_int* 0x7e864484)
[VC4CL](      simple_cl): Tracking live-time of object: 0x1e40114 (cl_context)
2: context[0x1e40114]
[VC4CL](      simple_cl): API call: cl_command_queue clCreateCommandQueue(cl_context 0x1e40114, cl_device_id 0x1e3b888, cl_command_queue_properties 0, cl_int* 0x7e864484)
[VC4CL](      simple_cl): Starting queue handler thread...
[VC4CL](      simple_cl): Tracking live-time of object: 0x1e3facc (cl_command_queue)
3: queue[0x1e3facc]
[VC4CL](      simple_cl): API call: cl_mem clCreateBuffer(cl_context 0x1e40114, cl_mem_flags 1, size_t 128, void* 0, cl_int* 0x7e864484)
[VC4CL](      simple_cl): [VC4CL] Mailbox file descriptor opened: 4
[VC4CL](      simple_cl): Mailbox buffer before: 00000020 00000000 00030012 00000008 00000004 00000001 00000000 00000000
[VC4CL](      simple_cl): Mailbox buffer after: 00000020 80000000 00030012 00000008 80000004 80000000 00000000 00000000
[VC4CL](      simple_cl): Mailbox buffer before: 00000020 00000000 00010006 00000008 00000000 00000000 00000000 00000000
[VC4CL](      simple_cl): Mailbox buffer after: 00000020 80000000 00010006 00000008 80000008 38000000 08000000 00000000
[VC4CL](      simple_cl): Mailbox request: succeeded
[VC4CL](      simple_cl): Tracking live-time of object: 0x1e13544 (cl_mem)
[VC4CL](      simple_cl): Mailbox buffer before: 00000024 00000000 0003000c 0000000c 0000000c 00000080 00001000 0000000c 00000000
[VC4CL](      simple_cl): Mailbox buffer after: 00000024 80000000 0003000c 0000000c 80000004 00000009 00001000 0000000c 00000000
[VC4CL](      simple_cl): Mailbox buffer before: 00000020 00000000 0003000d 00000008 00000004 00000009 00000000 00000000
[VC4CL](      simple_cl): Mailbox buffer after: 00000020 80000000 0003000d 00000008 80000004 be535000 00000000 00000000
[VC4CL](      simple_cl): [VC4CL] base=0x3e535000, mem=0x76f4a000
[VC4CL](      simple_cl): Allocated 128 bytes of buffer: handle 9, device address 0xbe535000, host address 0x76f4a000
4: memobj[0x1e13544]
[VC4CL](      simple_cl): API call: cl_program clCreateProgramWithSource(cl_context 0x1e40114, cl_uint 1, const char** 0x7e8643f4, const size_t* 0x7e8643f0, cl_int* 0x7e864484)
[VC4CL](      simple_cl): Tracking live-time of object: 0x1e1386c (cl_program)
5: program[0x1e1386c], source_size[305]
[VC4CL](      simple_cl): API call: cl_int clBuildProgram(cl_program 0x1e1386c, cl_uint 1, const cl_device_id* 0x7e864498, const char* (null), void(CL_CALLBACK*)(cl_program progr
am, void* user_data) 0x7e864324, void* 0)
[VC4CL](      simple_cl): Precompiling source with:
Dumping program sources to /tmp/vc4cl-source-1804289383.cl
[VC4CL](      simple_cl): Dumping program IR to /tmp/vc4cl-ir-846930886.ll
[VC4CL](      simple_cl): Precompilation complete with status: 0
[VC4CL](      simple_cl): [VC4CL] base=0x3fc00000, mem=0x76f49000
[VC4CL](      simple_cl): [VC4CL] V3D base: 0x76f49000
[VC4CL](      simple_cl): Compiling source with:
[VC4CL](      simple_cl): Compilation complete with status: 0
Dumping program binaries to /tmp/vc4cl-binary-1681692777.bin
6: build program ret[0]
[VC4CL](      simple_cl): API call: cl_kernel clCreateKernel(cl_program 0x1e1386c, const char* "hello", cl_int* 0x7e864484)
[VC4CL](      simple_cl): Tracking live-time of object: 0x1e195f4 (cl_kernel)
7: create kernel[0]
[VC4CL](      simple_cl): API call: cl_int clSetKernelArg(cl_kernel 0x1e195f4, cl_uint 0, size_t 4, const void* 0x7e864494)
[VC4CL](      simple_cl): Set kernel arg 0 for kernel 'hello' to 0x7e864494 (31536452) with size 4
[VC4CL](      simple_cl): Kernel arg 0 for kernel 'hello' is char* 'string' with size 4
[VC4CL](      simple_cl): Setting kernel-argument 0 to pointer 0x0x1e13538
8: clSetKernelArg ret[0]
[VC4CL](      simple_cl): API call: cl_int clEnqueueTask(cl_command_queue 0x1e3facc, cl_kernel 0x1e195f4, cl_uint 0, const cl_event* 0, cl_event* 0)
[VC4CL](      simple_cl): API call: cl_int clEnqueueNDRangeKernel(cl_command_queue 0x1e3facc, cl_kernel 0x1e195f4, cl_uint 1, const size_t* 0, const size_t* 0x7e864364, const siz
[... 40 lines cut ...]
[E] Thu Jan  1 09:19:51 1970:  (5) /lib/libstdc++.so.6 : __gnu_cxx::__verbose_terminate_handler()+0x93 [0x767c6098]
[E] Thu Jan  1 09:19:51 1970:  (6) /lib/libstdc++.so.6 : +0x8398c [0x767c498c]
[VC4CL](VC4CL Queue Han): [VC4CL] Leaked object 0x1e40114 with 6 references: cl_context
[VC4CL] Leaked object 0x1e3facc with 3 references: cl_command_queue
[VC4CL] Leaked object 0x1e13544 with 2 references: cl_mem
[VC4CL] Leaked object 0x1e1386c with 2 references: cl_program
[VC4CL] Leaked object 0x1e195f4 with 2 references: cl_kernel
[VC4CL] Leaked object 0x1e492ec with 2 references: cl_event
[VC4CL] Leaked object 0x1e42b74 with 3 references: cl_event

[VC4CL](VC4CL Queue Han): Mailbox buffer before: 00000020 00000000 0003000e 00000008 00000004 00000009 00000000 00000000
[VC4CL](VC4CL Queue Han): ioctl_set_msg failed: -1
[VC4CL] Error in mbox_property: Connection timed out
terminate called recursively
[E] Thu Jan  1 09:19:52 1970: Received signal: SIGABRT
[E] Thu Jan  1 09:19:52 1970:  (1) /lib/libVC4CC.so.1.2 : +0xdd9f0 [0x7698a9f0]
[E] Thu Jan  1 09:19:52 1970:  (2) /lib/libc.so.6 : __default_rt_sa_restorer+0 [0x76dd43a0]
[E] Thu Jan  1 09:19:52 1970:  (3) /lib/libc.so.6 : gsignal+0x9c [0x76dd34fc]
[E] Thu Jan  1 09:19:52 1970:  (4) /lib/libc.so.6 : abort+0x114 [0x76dbd788]
[E] Thu Jan  1 09:19:52 1970:  (5) /lib/libstdc++.so.6 : __gnu_cxx::__verbose_terminate_handler()+0xa9 [0x767c60ae]
[E] Thu Jan  1 09:19:52 1970:  (6) /lib/libstdc++.so.6 : +0x8398c [0x767c498c]
[ 1196.393172] raspberrypi-exp-gpio soc:firmware:expgpio: Failed to get GPIO 4 state (-110 84)
`

If I disable enlightenment(display manager), then run this sample, everything will be ok, we use mesa vc4 for Tizen RPI3, I'm wondering if  VC4CL can't run simultaneously with mesa?

doe300, 2020-11-09:
VC4CL is not compatible with the "full kms" OpenGL driver, see also #60.

To clarify: You can run VC4CL while having a GUI (and OpenGL support) open, just not with the "full kms" driver and also probably not running an OpenGL application and an OpenCL application at the same time.

xuelian-bai, 2020-11-10:
ok, I see, thanks

closed 2020-11-10 by xuelian-bai