doe300/VC4CL #104: [Question] Is VC4CL compatible with BOINC GPU jobs ?

sviscapi, 2021-12-26:
Dear @doe300 ,

I found out only recently about your OpenCL implementation. I had a RPI 3 lying idle somewhere so I decided to put it to good use and make it crunch for science with BOINC (Einstein@Home project, to be precise):

https://einsteinathome.org/fr/home

It takes roughly 14 hours for the RPI 3 to complete a single CPU job. So far I haven't been able to contribute much horse power to the project...

Could I use VC4CL to run GPU jobs on my RPI 3 and improve my results ?

Best regards,

Samuel

doe300, 2021-12-27:
Although BOINC does support OpenCL to some degree, having a quick look at the Einstein@Home source code, I can't see any use of it.
That said, it might work, although it is likely to not be very fast and/or produce wrong results.

sviscapi, 2021-12-27:
Dear @doe300 ,

Thank you for your quick reply, really appreciated.

According to this page (https://einsteinathome.org/fr/application-source-code-and-license), the application currently running on my RPI 3 (Binary Radio Pulsar Search) is compatible with OpenCL:

> You can use the script build.sh to build native binaries for Windows (cross-compiled on Linux using MinGW!), Mac OS X and Linux. This script will download all 3rd-party libraries and build them before building the actual science application. The script will check whether all prerequisites are met and supports checkpointing. For CUDA or OpenCL builds, please make sure PATH as well as CUDA_INSTALL_PATH or AMDAPPSDKROOT respectively are set properly. 

There's indeed an OpenCL directory in the source archive: https://einsteinathome.org/brp-src-release.zip.

This is my first foray into GPGPU and I must say this is all very confusing :)

Best regards,

Samuel

doe300, 2021-12-28:
From what I can see from the source code, the application requires OpenCL 1.1, which VC4CL implements. It also seems to be able to dynamically use smaller work-group sizes (VC4CL has a limit of 12).

Currently some kernels seem to not compile for VC4CL, so it won't be (fully) usable right now.

ssb22, 2021-12-28:
Don't forget VC4CL supports only the Embedded profile (i.e. OpenCL EP), not the full OpenCL

sviscapi, 2022-01-08:
Dear @doe300 and @ssb22 ,

I went ahead and posted my question at the Einstein@Home boards:

https://einsteinathome.org/fr/content/opencl-requirements-eh-arm-platform#comment-191610

Looks like I'm out of luck (for now).

Best regards,

Samuel