doe300/VC4CL #107: Tests fail for Raspberry Pi Zero

Meta-chan, 2022-02-08:
Hello. I'm trying to run OpenCL on Raspberry Pi Zero with Raspbian (Debian Buster). I know that the library was not designed for Zero, but since [this guy](https://kenny-peng.com/2021/09/14/raspi_zero_opencl.html) claimed it worked for him, I decided to give it a try. So here is what I discovered:

1) `VC4CL` cannot be built on latest Raspbian, because it doesn't have `/opt/vc/`. Not sure if it is supposed to be there or they moved it to other place.
2) Latest version of `VC4C` throws `unexpected reloc type 0x03` error. Could be fixed with '-DCMAKE_CXX_FLAGS=-fPIC'.
3) `VC4CL` commit with "stable" tag cannot be compiled: `Program.cpp` needs `Optional.h` but can't find it. Can be fixed easily.
4) `TestVC4CL` from 2021/09/14 passes only 36.6% of tests.
5) After playing with `/opt/vc/bin/container_test` I discovered that now `TestVC4CL` passes 54% of tests.

Could you please comment this situation?
Even if it doesn't work for me, you do a great job. Thank you.

doe300, 2022-02-08:
1. a60945c9982a8ed07b7c216a308bf4beee903bce (or just use latest master) should fix it. If it does not, can you give me the exact thing which fails + message?
2. see https://github.com/doe300/VC4CL/issues/105#issuecomment-1024877957, I should probably disable by default or monkey-patch the validator
3. was my fault, I restructured code in VC4C too soon ;) Both masters should work well together.
4. I am currently working on improving test stability for both VC4C and VC4CL, but some of the tests for TestVC4CL (or TestVC4C for that matter) never passed. I guess these are less in a sense of regression tests, more like to have some code I can test (and fix).
5. What is `/opt/vc/bin/container_test`, I don't think I have ever heard of it? What did you change there?

About the test passage, VC4CL is not complete, so while I agree having failing own tests may be confusing, other tests won't pass 100% too. E.g. currently it passes around 90+% of [boost/compute](https://github.com/boostorg/compute) tests, some only after fixing or modifying the tests to work on the platform and IIRC about 2/3 of OpenCL [piglit](https://piglit.freedesktop.org/) tests. Similarly the [OpenCL-CTS](https://github.com/KhronosGroup/OpenCL-CTS/) tests are largely passed, but with some major holes (e.g. anything non-trival math related).

Meta-chan, 2022-02-08:
1) No it does not, because there still is a line `elseif(CROSS_COMPILE OR EXISTS "/opt/vc/include/bcm_host.h")`. It doesn't see `/opt/vc`. This would probably result in `undefined reference to bcm_host_init` (but I need a day to recompile everyting)
2,3) Happens. Just saying
4) Wouldn't it be better to create simple "2 + 2" example to verify if the library is working or not? Everyone says about boost tests and 90% of support. I don't know much about boost, I am interested if my program will work or not. I need to verify I have right compiler, right kernel, right firmware and right everything. And when I see "clCompileProgram" I get a feeling that something has gone terribly wrong and the whole thing doesn't work.
5) Well, I had the idea that maybe official programs will be executed without problems, so I played with programs from `/opt/vc/bin`. And apparently `container_test` made something to GPU so that tests executed better (I also noticed that pseudographics in raspi-config were replaced by characters).
6) Okay, I finally brought it to work, and I want to say that VC4CL is extremely unfriendly. `CL/ocl.h` has functions with `cl` prefix, while VC4CL has `VC4CL_` prefix. So functions are compiled, but cannot be linked. What I did is including "icd_loader.h" and adding "-Duse_cl_khr_icd=true", only then it could be executed.
And when I execute [this example](https://github.com/rsnemmen/OpenCL-examples/blob/master/Hello_World/hello.c) (that you mentioned as a helloworld in [this topic](https://github.com/doe300/VC4CL/issues/46) ), I get `clEnqueueNDRangeKernel` fail.
All of this still applies to source from  2021/09/14. Should I build the latest?

doe300, 2022-02-08:
Oh right, missed that one. Does the header file `/usr/include/bcm_host.h` exist on your system? Or where the is the header `bcm_host.h` located?

> I don't know much about boost, I am interested if my program will work or not.

Not sure if I understand you correctly, but if you want to run own code then I cannot guarantee that it works, since that depends on so many things. You will have to try it out and if the compilation fails then it could very well be a bug/missing feature in VC4C.

The issue with the prefixes is that there are basically 2 ways of using any OpenCL implementation: directly or via an ICD loader. If you want to use it directly, configure CMake with `-DBUILD_ICD=OFF`, then the `clXXX` symbols will be provided. If used via an ICD loader, then the loader will do the function mapping depending on the active implementation. See also https://github.com/doe300/VC4CL#khronos-icd-loader for how to enable ICD loader support for a source-build.

As for the source code version, I would always recommend to use latest master.

Meta-chan, 2022-02-08:
> Does the header file `/usr/include/bcm_host.h` exist on your system? Or where the is the header `bcm_host.h` located?
I reinstalled my system, I have Buster now. there is no `/usr/include/bcm_host.h`, I have it in `/opt/vc/include/bcm_host.h`. I guess same applied to Bullseye, but not 100% sure.

> Not sure if I understand you correctly, but if you want to run own code then I cannot guarantee that it works, since that depends on so many things.
Well, it would be nice if you could provide some simple program and say "This one is one true way. If it is compiled and works, other programs written in similar way may probably also work. If it is not compiled or doesn't work, it's your problem". Instead you provide no examples and don't guarantee that user's code works. Totally fair from your side, but it doesn't help anybody in using the library.

Thanks for `icd-loader` advice. As said, I'll compile the newest version till tomorrow evening and might update some info.

doe300, 2022-02-09:
About which software should work: Any OpenCL client application which works on other OpenCL implementations and does not use some implementation specific features or unsupported extensions shoudl work. If it does not, it is likely a bug in VC4CL.

I just tested on a fresh installation of Bullseye and besides one fix with the header paths (which I will probably commit today or so) it compiles fine. But Buster is definitively tested more extensively.

Meta-chan, 2022-02-09:
Okay, I tried [this example](https://github.com/rsnemmen/OpenCL-examples/tree/master/add_numbers), started it with `sudo` and it works. You are a genius, man. You don't accept donations, do you?