doe300/VC4CL #108: Support Raspberry OS 64 bit Bullseye

cantricao, 2022-04-20:
In 64 bit os, raspberry do not have libvcsm.so.
I edited file to delete all code related to vcsm and build successfully.
Could you add 64 bit flag to omit the VC4CL_MEMORY_VCSM and VC4CL_MEMORY_CMA?
Thank you.

doe300, 2022-04-21:
You could set the `VC4CL_NO_VCSM` environment variable, which will skip the usage of the libvcsm.so.

cantricao, 2022-04-21:
Yes, that I want to say.
Another thing I noticed is that vc4cl must have sudo permission when that flag turn on.

illwieckz, 2022-11-03:
I'm also trying to run VC4CL on 64bit Raspberry Pi (Pi 3b+), do I miss something?

```
$ sudo bash -c 'VC4CL_NO_VCSM=1 OCL_ICD_VENDORS=/opt/vc4cl/lib/libVC4CL.so clinfo --list'
Bus error
```

illwieckz, 2022-11-03:
I don't get something that looks much useful:

```
$ sudo bash -c 'VC4CL_DEBUG=system VC4CL_NO_VCSM=1 OCL_ICD_VENDORS=/opt/vc4cl/lib/libVC4CL.so gdb -iex "set pagination off" -ex "run" -ex "thread apply all backtrace" -args clinfo --list'
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from clinfo...
(No debugging symbols found in clinfo)
Starting program: /usr/bin/clinfo --list
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".

Program received signal SIGBUS, Bus error.
0x0063697665645f61 in ?? ()

Thread 1 (Thread 0x7ff7ff6040 (LWP 31643) "clinfo"):
#0  0x0063697665645f61 in ?? ()
#1  0x0000007ff7d108c0 in clGetExtensionFunctionAddress (name=0x7ff7fa3230 "clIcdGetPlatformIDsKHR") at ../src/extensions.cpp:53
#2  0x0000007ff7f9ebf4 in ?? () from /lib/aarch64-linux-gnu/libOpenCL.so.1
#3  0x0000007ff7f9f73c in clGetPlatformIDs () from /lib/aarch64-linux-gnu/libOpenCL.so.1
#4  0x0000005555557074 in ?? ()
#5  0x0000007ff7e30e18 in __libc_start_main (main=0x5555556f00, argc=2, argv=0x7ffffff4e8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=<optimized out>) at ../csu/libc-start.c:308
#6  0x0000005555557620 in ?? ()
Backtrace stopped: not enough registers or memory available to unwind further
```