doe300/VC4CL #113: No way to build with cpplog, also it does not find installed cpplog

illwieckz, 2022-10-27:
With default source I get:

```
ninja: error: 'cpplog/src/cpplog-project-build/libcpplog-static.a', needed by 'src/libVC4CC.so.0.4.9999',
 missing and no known rule to make it
```

With this patch:

```diff
diff --git a/cmake/cpplog.cmake b/cmake/cpplog.cmake
index 84b5245..41fcced 100644
--- a/cmake/cpplog.cmake
+++ b/cmake/cpplog.cmake
@@ -1,4 +1,4 @@
-if(FALSE AND DEPENDENCIES_USE_FETCH_CONTENT)
+if(DEPENDENCIES_USE_FETCH_CONTENT)
        # TODO disabled for now, since cpplog uninstall target conflicts with VC4C uninstall target
        include(FetchContent)
        FetchContent_Declare(cpplog GIT_REPOSITORY https://github.com/doe300/cpplog.git GIT_TAG v0.6)
```

I get:

```
CMake Error at src/CMakeLists.txt:146 (add_custom_target):
  add_custom_target cannot create target "uninstall" because another target
  with the same name already exists.  The existing target is a custom target
  created in source directory "VC4C/build/_deps/cpplog-src".
  See documentation for policy CMP0002 for more details.
```

If I install cpplog in the install prefix as an alternative way to build and provide it, CMake doesn't find it.

I guess either making the cpplog install step conditional to not being part of a parent project, either making vc4cl finds installed cpplog, would fix that. Right now both building as part of vc4cl or separately can't make vc4cl build.

illwieckz, 2022-11-02:
It looks like this error is specific to Ninja:

```
ninja: error: 'cpplog/src/cpplog-project-build/libcpplog-static.a', needed by 'src/libVC4CC.so.0.4.9999',
 missing and no known rule to make it
```

When I run cmake with `-G'Unix Makefiles'` it continues.

Anyway this is an issue for [VC4C](https://github.com/doe300/VC4C), I reported it on the wrong repo, this may me moved to VC4C.

doe300, 2022-11-03:
I never built the projects with Ninja. From the error it looks like some dependency is not known to or correctly handled by CMake...
If this error happens under Make, then simply running make twice should also fix it.

illwieckz, 2022-11-03:
Yeah, basically the problem seems to be that Ninja expects the `libcpplog-static.a` target to be built before starting to built, while it is built as part of the build process…