doe300/VC4CL #114: issues when building tools and tests

psmatter, 2023-02-10:
When linking the v3d_info tool, the linker complains with unresolved externals. 

Patching  -Wl,--no-as-needed -ldl -pthread manually into build/tools/CMakeFiles/v3d_info.dir/link.txt  solves the issue. 

When building the tests, TestData.h is missing. 

(using a raspi running bullseye)

doe300, 2023-02-10:
What are the externals that are not found?

psmatter, 2023-02-10:
When building Tests from add_subdirectory(test) in the CMakefile.txt,   test.cpp and TestExecutions.cpp miss TestData.h

psmatter, 2023-02-10:
Sorry, you asked for the unresolved externals. I didnt note them. pthread_create as far as I remember and dlopen to dlsym.

psmatter, 2023-02-10:
Just cloned the repository and compiled again, but now everything compiles fine. Might eventually come from me compiling VC4CL prior to VC4CLStdLib and VC4C.

psmatter, 2023-02-10:
Just confirmed. TestData.h is in the VC4C/test directory. Also the unresolved externals are gone. Was def the wrong way to start with VC4CL. Up and running now:
clinfo -l
Platform #0: OpenCL for the Raspberry Pi VideoCore IV GPU
   Device #0: VideoCore IV GPU

Thanks a lot and sorry for the inconvenience!

closed 2023-02-10 by psmatter