doe300/VC4CL #115: Failed to download dependencies when compiling VC4C

Zenitur, 2023-04-17:
```
pi@raspberrypi:~/opencl-vc4 $ cd VC4C
pi@raspberrypi:~/opencl-vc4/VC4C $ ls
check_all_files.sh  collect_coverage.sh  include    scripts  testing
cmake               doc                  LICENSE    src      tools
CMakeLists.txt      example              Readme.md  test
pi@raspberrypi:~/opencl-vc4/VC4C $ mkdir build && cd build
pi@raspberrypi:~/opencl-vc4/VC4C/build $ cmake ..
CMake Deprecation Warning at CMakeLists.txt:4 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- VC4CL standard library headers found: /home/pi/opencl-vc4/VC4C/../VC4CLStdLib/include/
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CMake 3.14+ FetchContent to include dependencies...
Scanning dependencies of target spirv-headers-populate
[ 11%] Creating directories for 'spirv-headers-populate'
[ 22%] Performing download step (git clone) for 'spirv-headers-populate'
\u041a\u043b\u043e\u043d\u0438\u0440\u043e\u0432\u0430\u043d\u0438\u0435 \u0432 \u00abspirv-headers-src\u00bb\u2026
fatal: \u043d\u0435\u043f\u0440\u0430\u0432\u0438\u043b\u044c\u043d\u0430\u044f \u0441\u0441\u044b\u043b\u043a\u0430: master
CMake Error at spirv-headers-subbuild/spirv-headers-populate-prefix/tmp/spirv-headers-populate-gitclone.cmake:40 (message):
  Failed to checkout tag: 'master'

make[2]: *** [CMakeFiles/spirv-headers-populate.dir/build.make:92: spirv-headers-populate-prefix/src/spirv-headers-populate-stamp/spirv-headers-populate-download] \u041e\u0448\u0438\u0431\u043a\u0430 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/spirv-headers-populate.dir/all] \u041e\u0448\u0438\u0431\u043a\u0430 2
make: *** [Makefile:84: all] \u041e\u0448\u0438\u0431\u043a\u0430 2

CMake Error at /usr/share/cmake-3.16/Modules/FetchContent.cmake:915 (message):
  Build step for spirv-headers failed: 2
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FetchContent.cmake:1006 (__FetchContent_directPopulate)
  /usr/share/cmake-3.16/Modules/FetchContent.cmake:1047 (FetchContent_Populate)
  cmake/spirv-headers.cmake:9 (FetchContent_MakeAvailable)
  CMakeLists.txt:110 (include)

-- Configuring incomplete, errors occurred!
See also "/home/pi/opencl-vc4/VC4C/build/CMakeFiles/CMakeOutput.log".

```

Zenitur, 2023-04-22:
Yeah, that was a fix.

doe300, 2023-05-30:
Fixed by https://github.com/doe300/VC4C/pull/164

closed 2023-05-30 by doe300