doe300/VC4CL #116: OpenCV DNN Module: OpenCL Target Precompilation Error

angryGoat500, 2023-08-10:
I have an object detection program running in OpenCV DNN Module.
When I configure the OpenCL Backend in my code:
`  net = cv2.dnn.readNetFromONNX(modelPath)
  net.setPreferableBackend(3)
  net.setPreferableTarget(1)
`
I get all of these Precompilation Errors in VC4C / VC4CL? (heavily shortened)
The clinfo output is posted at the end.

What am I doing wrong here? Do I need to change my code? Has the installation of VC4C / VC4CL not been successful?

Thanks in advance!

2023-08-10 23:14:23,751 __main__ INFO Hallo
OpenCL program build log: dnn/dummy
Status -15: CL_COMPILE_PROGRAM_FAILURE
-cl-no-subgroup-ifp
Precompilation error:
	Pre-compilation: Error in precompilation: error: unknown argument: '-cl-no-subgroup-ifp'

[E] Thu Aug 10 23:14:33 2023: Errors in precompilation for: /usr/bin/clang -cc1 -triple spir-unknown-unknown -cl-no-subgroup-ifp -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant -fgnu89-inline -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -finclude-default-header -include /root/opencl/VC4C/../VC4CLStdLib/include/defines.h -x cl -S -emit-pch -o /tmp/vc4c-openclc-pch-LwkcGp -
[E] Thu Aug 10 23:14:33 2023: error: unknown argument: '-cl-no-subgroup-ifp'

[E] Thu Aug 10 23:14:33 2023:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0x62638ce4]
[E] Thu Aug 10 23:14:33 2023:  (2) /usr/local/lib/libVC4CC.so.1.2 : +0xf639a0 [0x62b419a0]
[E] Thu Aug 10 23:14:33 2023:  (3) /usr/local/lib/libVC4CC.so.1.2 : +0xf6b43c [0x62b4943c]
[E] Thu Aug 10 23:14:33 2023:  (4) /usr/local/lib/libVC4CC.so.1.2 : +0xf64b04 [0x62b42b04]
[E] Thu Aug 10 23:14:33 2023:  (5) /usr/local/lib/libVC4CC.so.1.2 : +0xf64e8c [0x62b42e8c]
[E] Thu Aug 10 23:14:33 2023:  (6) /usr/local/lib/libVC4CC.so.1.2 : +0xf685e0 [0x62b465e0]
[E] Thu Aug 10 23:14:33 2023:  (7) /usr/local/lib/libVC4CC.so.1.2 : vc4c::precompilation::compileOpenCLToLLVMIR(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&&)+0xc4 [0x62b4676c]
[E] Thu Aug 10 23:14:33 2023:  (8) /usr/local/lib/libVC4CC.so.1.2 : +0xf9bf80 [0x62b79f80]
[E] Thu Aug 10 23:14:33 2023:  (9) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::precompile(vc4c::CompilationData const&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xd8 [0x62b77a64]
[E] Thu Aug 10 23:14:33 2023:  (10) /usr/local/lib/libVC4CL.so : +0x18047c [0x62f0147c]
[E] Thu Aug 10 23:14:33 2023:  (11) /usr/local/lib/libVC4CL.so : vc4cl::Program::compile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, vc4cl::object_wrapper<vc4cl::Program>, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, vc4cl::object_wrapper<vc4cl::Program> > > > const&)+0x108 [0x62f034ac]
[E] Thu Aug 10 23:14:33 2023:  (12) /usr/local/lib/libVC4CL.so : +0x183794 [0x62f04794]
[E] Thu Aug 10 23:14:33 2023:  (13) /usr/local/lib/libVC4CL.so : VC4CL_clBuildProgram+0x40c [0x62f06524]
[E] Thu Aug 10 23:14:33 2023:  (14) /usr/local/lib/libopencv_core.so.408 : +0x1a3d74 [0x6b7e1d74]
[E] Thu Aug 10 23:14:33 2023:  (15) /usr/local/lib/libopencv_core.so.408 : +0x1ac518 [0x6b7ea518]
[E] Thu Aug 10 23:14:33 2023:  (16) /usr/local/lib/libopencv_core.so.408 : +0x1bb770 [0x6b7f9770]
[E] Thu Aug 10 23:14:33 2023:  (17) /usr/local/lib/libopencv_core.so.408 : cv::ocl::Program::create(cv::ocl::ProgramSource const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)+0x68 [0x6b7f9eec]
[E] Thu Aug 10 23:14:33 2023:  (18) /usr/local/lib/libopencv_core.so.408 : cv::ocl::Program::Program(cv::ocl::ProgramSource const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)+0x14 [0x6b7fa058]
[E] Thu Aug 10 23:14:33 2023:  (19) /usr/local/lib/libopencv_core.so.408 : cv::ocl::Context::Impl::getProg(cv::ocl::ProgramSource const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)+0x2f8 [0x6b7fa358]
[E] Thu Aug 10 23:14:33 2023:  (20) /usr/local/lib/libopencv_core.so.408 : cv::ocl::Context::getProg(cv::ocl::ProgramSource const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)+0x24 [0x6b7fa648]
[E] Thu Aug 10 23:14:33 2023:  (21) /usr/local/lib/libopencv_dnn.so.408 : +0x2ed1fc [0x6c2c41fc]
[E] Thu Aug 10 23:14:33 2023:  (22) /usr/local/lib/libopencv_dnn.so.408 : +0x2f3ab0 [0x6c2caab0]
[E] Thu Aug 10 23:14:33 2023:  (23) /usr/local/lib/libopencv_dnn.so.408 : +0x2f43b4 [0x6c2cb3b4]
[E] Thu Aug 10 23:14:33 2023:  (24) /usr/local/lib/libopencv_dnn.so.408 : +0x2fa5f8 [0x6c2d15f8]
[E] Thu Aug 10 23:14:33 2023:  (25) /usr/local/lib/libopencv_dnn.so.408 : +0x2fa854 [0x6c2d1854]
[E] Thu Aug 10 23:14:33 2023:  (26) /usr/local/lib/libopencv_dnn.so.408 : +0x2ffcac [0x6c2d6cac]
[E] Thu Aug 10 23:14:33 2023:  (27) /usr/local/lib/libopencv_dnn.so.408 : +0x2fffc4 [0x6c2d6fc4]
[E] Thu Aug 10 23:14:33 2023:  (28) /usr/local/lib/libopencv_dnn.so.408 : +0x155714 [0x6c12c714]
[E] Thu Aug 10 23:14:33 2023:  (29) /usr/local/lib/libopencv_dnn.so.408 : +0x157778 [0x6c12e778]
[E] Thu Aug 10 23:14:33 2023:  (30) /usr/local/lib/libopencv_dnn.so.408 : cv::dnn::dnn4_v20230620::Net::Impl::forwardLayer(cv::dnn::dnn4_v20230620::detail::LayerData&)+0x6e0 [0x6c2909e0]
[E] Thu Aug 10 23:14:33 2023:  (31) /usr/local/lib/libopencv_dnn.so.408 : cv::dnn::dnn4_v20230620::Net::Impl::forwardToLayer(cv::dnn::dnn4_v20230620::detail::LayerData&, bool)+0x88 [0x6c28d540]
[E] Thu Aug 10 23:14:33 2023:  (32) /usr/local/lib/libopencv_dnn.so.408 : cv::dnn::dnn4_v20230620::Net::Impl::forward(cv::_OutputArray const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)+0x108 [0x6c2a7244]
[E] Thu Aug 10 23:14:33 2023:  (33) /usr/local/lib/libopencv_dnn.so.408 : cv::dnn::dnn4_v20230620::Net::forward(cv::_OutputArray const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)+0x54 [0x6c287220]
[E] Thu Aug 10 23:14:33 2023:  (34) /usr/local/lib/python3.9/dist-packages/cv2/python-3.9/cv2.so : +0x3456a0 [0x6d1206a0]
[E] Thu Aug 10 23:14:34 2023: Errors in precompilation for: /usr/bin/clang -cc1 -triple spir-unknown-unknown -cl-no-subgroup-ifp -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant -fgnu89-inline -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -finclude-default-header -include /root/opencl/VC4C/../VC4CLStdLib/include/defines.h -x cl -S -emit-llvm-bc -o /tmp/vc4c-ousIxn -
[E] Thu Aug 10 23:14:34 2023: error: unknown argument: '-cl-no-subgroup-ifp'

[E] Thu Aug 10 23:14:34 2023:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0x62638ce4]
[E] Thu Aug 10 23:14:34 2023:  (2) /usr/local/lib/libVC4CC.so.1.2 : +0xf639a0 [0x62b419a0]
[E] Thu Aug 10 23:14:34 2023:  (3) /usr/local/lib/libVC4CC.so.1.2 : +0xf6ad64 [0x62b48d64]
[... cut ...]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 OpenCL for the Raspberry Pi VideoCore IV GPU
    Device Name                                   VideoCore IV GPU

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.14
  ICD loader Profile                              OpenCL 3.0

doe300, 2023-08-11:
The error is (reported from the clang front-end):

> error: unknown argument: '-cl-no-subgroup-ifp'

According to the OpenCL specification this command line option was added in OpenCL 2.1, which is not supported by VC4CL.

angryGoat500, 2023-08-12:
Thanks for your reply! 

I assume it means I need to fix my clang somehow? How would I do that?
Do I need o rebuild  the VC4CL Binaries with `--ffast-math flag` as suggested in https://github.com/opencv/opencv/issues/13852 ?

How would I do that?

And how could I enable the NEON optimization?

doe300, 2023-08-14:
> I assume it means I need to fix my clang somehow? How would I do that?

No, the error means that OpenCV uses an OpenCL 2.1+ feature on an OpenCL 1.2 implementation, so it looks like the error is on OpenCV side. No updating of clang or rebuilding of VC4CL can fix this.

> Do I need o rebuild the VC4CL Binaries with --ffast-math flag as suggested in https://github.com/opencv/opencv/issues/13852 ?

You probably don't want to rebuild VC4CL with that flag (that would probably not work and even if it did, not do anything), you may want to rebuild the VC4CL standard library prebuilt with the flag. This can be done by adapting the argument lists in https://github.com/doe300/VC4C/blob/master/src/precompilation/FrontendCompiler.cpp#L917 ff.

> And how could I enable the NEON optimization?

VC4CL does not use NEON (except maybe from compiler-generated vectorizations), so this does not apply here. The question is probably about the OpenCV CPU backend, so you'll  need to look there.

closed 2023-08-14 by angryGoat500