doe300/VC4CL #118: Running on Docker

Dall127, 2023-10-26:
Is there a kind soul that has used this to get video acceleration on a docker image? I'm currently working on trying to get hardware acceleration for Obico's 3D printing machine learning model. Ive got a darknet build that supports opencl and specifically will run on the raspberry pi via this package, but every time I've tried to bring up the GPU, I've been getting **CL_INVALID_CONTEXT**. My best hypothesis is that I'm not passing through the correct devices in the docker image, or incorrectly configured settings on the raspberry pi (model 3a+) to bring up the gpu and make it accessible. Any help would be greatly appreciated!

doe300, 2023-10-26:
VC4CL directly accesses hardware registers and memory-maps GPU memory, so I doubt it will run in any container.
But at least it will need access to (some of the) following:

- `/dev/vcsm-cma` or `/dev/vcsm`
- `/dev/vcio`
- `/dev/mem`

Dall127, 2023-10-26:
Thank you so much! I believe that docker has a mode called privileged that gives lower level access to the hardware, so I'll map those devices and set that flag and give it a try. I'll report back to this issue on the success of it. Thank you again!