doe300/VC4CL #119: Could not find supported libbcm_host.so

jskimn, 2024-02-19:
-----------------------------------------------------------------------------------------------------------------------------
OS: PRETTY_NAME="Debian GNU/Linux 11 (bullseye)" 64bit
Refrenced Issues: https://github.com/doe300/VC4CL/issues/108, https://github.com/doe300/VC4CL/issues/105
-----------------------------------------------------------------------------------------------------------------------------

Thx for your open source.
I don't know why following problem is occured.
And I checked issue #108 and #105, but I could not find any reason.

$ clinfo
Number of platforms                               1
  Platform Name                                   OpenCL for the Raspberry Pi VideoCore IV GPU
  Platform Vendor                                 doe300
  Platform Version                                OpenCL 1.2 VC4CL 0.4.9999 (b5a6097)
  Platform Profile                                EMBEDDED_PROFILE
  Platform Extensions                             cl_khr_il_program cl_khr_spir cl_khr_create_command_queue cl_altera_device_temperature cl_altera_live_object_tracking cl_khr_icd cl_khr_extended_versioning cl_khr_spirv_no_integer_wrap_decoration cl_khr_suggested_local_work_size cl_vc4cl_performance_counters
  Platform Extensions function suffix             VC4CL

  Platform Name                                   OpenCL for the Raspberry Pi VideoCore IV GPU
Number of devices                                 1
  Device Name                                     VideoCore IV GPU
  Device Vendor                                   Broadcom
  Device Vendor ID                                0x14e4
  Device Version                                  OpenCL 1.2 VC4CL 0.4.9999 (b5a6097)
  Device Numeric Version                          0x402000 (1.2.0)
  Driver Version                                  0.4.9999
  Device OpenCL C Version                         OpenCL C 1.2
  Device Type                                     GPU
  Device Profile                                  EMBEDDED_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Max compute units                               1
  Available core IDs                              0
terminate called after throwing an instance of 'std::system_error'
  what():  Could not find supported libbcm_host.so: No such file or directory
[E] Mon Feb 19 05:28:41 2024: Received signal: SIGABRT
[E] Mon Feb 19 05:28:41 2024:  (1) /usr/local/lib/libVC4CC.so.1.2 : +0xc52a04 [0x7f8c11ba04]
[E] Mon Feb 19 05:28:41 2024:  (2) linux-vdso.so.1 : __kernel_rt_sigreturn+0 [0x7f8cd187a8]
[E] Mon Feb 19 05:28:41 2024:  (3) /lib/aarch64-linux-gnu/libc.so.6 : gsignal+0xdc [0x7f8cb74eac]
[E] Mon Feb 19 05:28:41 2024:  (4) /lib/aarch64-linux-gnu/libc.so.6 : abort+0x108 [0x7f8cb61aa0]
[E] Mon Feb 19 05:28:41 2024:  (5) /lib/aarch64-linux-gnu/libstdc++.so.6 : __gnu_cxx::__verbose_terminate_handler()+0x188 [0x7f8b392238]
[E] Mon Feb 19 05:28:41 2024:  (6) /lib/aarch64-linux-gnu/libstdc++.so.6 : +0x9ed4c [0x7f8b38fd4c]
[E] Mon Feb 19 05:28:41 2024:  (7) /lib/aarch64-linux-gnu/libstdc++.so.6 : +0x9edb0 [0x7f8b38fdb0]
[E] Mon Feb 19 05:28:41 2024:  (8) /lib/aarch64-linux-gnu/libstdc++.so.6 : +0x9f0a4 [0x7f8b3900a4]
[E] Mon Feb 19 05:28:41 2024:  (9) /usr/local/lib/libVC4CL.so : +0x1d5ed0 [0x7f8cab4ed0]
[E] Mon Feb 19 05:28:41 2024:  (10) /usr/local/lib/libVC4CL.so : vchi_initialise+0x68 [0x7f8cab25ac]
[E] Mon Feb 19 05:28:41 2024:  (11) /usr/local/lib/libVC4CL.so : vc4cl::VCHI::VCHI()+0x18 [0x7f8cab8a74]
[E] Mon Feb 19 05:28:41 2024:  (12) /usr/local/lib/libVC4CL.so : +0x1cb604 [0x7f8caaa604]
[E] Mon Feb 19 05:28:41 2024:  (13) /usr/local/lib/libVC4CL.so : vc4cl::SystemAccess::SystemAccess()+0xcc [0x7f8caaa708]
[E] Mon Feb 19 05:28:41 2024:  (14) /usr/local/lib/libVC4CL.so : vc4cl::system()+0x70 [0x7f8caab8d0]
[E] Mon Feb 19 05:28:41 2024:  (15) /usr/local/lib/libVC4CL.so : vc4cl::Device::getInfo(unsigned int, unsigned long, void*, unsigned long*) const+0x290 [0x7f8ca28bb0]
[E] Mon Feb 19 05:28:41 2024:  (16) /usr/local/lib/libVC4CL.so : VC4CL_clGetDeviceInfo+0x13c [0x7f8ca29bfc]
[E] Mon Feb 19 05:28:41 2024:  (17) clinfo : +0xe7b4 [0x556403e7b4]
[E] Mon Feb 19 05:28:41 2024:  (18) clinfo : +0xe888 [0x556403e888]
[E] Mon Feb 19 05:28:41 2024:  (19) clinfo : +0x11940 [0x5564041940]
[E] Mon Feb 19 05:28:41 2024:  (20) clinfo : +0x1279c [0x556404279c]
[E] Mon Feb 19 05:28:41 2024:  (21) clinfo : +0x12bb4 [0x5564042bb4]
[E] Mon Feb 19 05:28:41 2024:  (22) clinfo : +0x728c [0x556403728c]
[E] Mon Feb 19 05:28:41 2024:  (23) /lib/aarch64-linux-gnu/libc.so.6 : __libc_start_main+0xe8 [0x7f8cb61e18]
[E] Mon Feb 19 05:28:41 2024:  (24) clinfo : +0x7620 [0x5564037620]

jskimn, 2024-02-19:
And My raspberry is 3 B model, it has libbcm_host library like below.

$ locate libbcm_host.so
/usr/lib/aarch64-linux-gnu/libbcm_host.so
/usr/lib/aarch64-linux-gnu/libbcm_host.so.0

doe300, 2024-02-19:
See https://github.com/doe300/VC4CL/blob/master/src/hal/userland.cpp#L42, the library path of your Debian OS is not listed, so you might need to add it manually.

jskimn, 2024-02-19:
Thanks for your answer. I will try to change using symbolic link or code change..
And then I will update for this issue.

jskimn, 2024-02-19:
I added aarch64 path in 'lookupPaths'.
I got clinfo result without any problem.
thanks @doe300.

closed 2024-02-19 by jskimn